eez-open/modular-psu #2: Length match

LJJS, 2018-08-29:
The length of the bus for the SDRam chip should have all the same length.
The bus operates at 143MHz, so the total trace length should be all equal.

Also, the length of each trace at each layer should have the same length!

prasimix, 2018-08-29:
Thanks for that. This is a first prototype and we'll see how it's going to work. I do believe that for keeping all traces of equal length is not ask for introducing 6-layer PCB.

LJJS, 2018-08-29:
I have done some testing with the length matching. Just an example how it can be done:

[image]

Sure, this is not the final solution, but I think it shows with some effort it is possible to tune the traces to very similar lengths. 
The board right now have differences in length to a maximum of 10mm, with a typical trace length of about 35mm. With some work it should be possible to match the length of the trace with a maximum difference of 1mm, even with a 4-layer PCB.

But also said, I think it will work just fine with these lengths at these frequencies, but it would be a nice detail in engineering.

prasimix, 2018-08-29:
Thanks for effort. Actually I tried to use meandering but the results was pathetic at least in v.7.4. I'm going to commit a fix for #3 and such changes should be applied in that revision.

Rerouter, 2018-09-08:
Most of the top layer can be matched without meandering, I have done it in kicad, but would probably be a pain to recreate in eagle. 
[image]
[image]

The bottom layer needs a bit more agressive length matching, currently i have matched within 10mm, 
I approached it by shortening the longest traces first, leaving almost nothing to be added to the top layers, 

But the control lines, some of them have 30mm differences, that took some manual creativity,

Rerouter, 2018-09-08:
LJJS, the timing specifications only require a +-10mm matching, Attached is my current effort to that 10mm margin, but if you want to continue down to under 1mm, be my guest. 

If you want to go the whole hog, the RX/TX buses on the ethernet controller are also out of skew by up to 30mm, 

[EEZ DIB MCU r1B1.zip](https://github.com/eez-open/modular-psu/files/2363171/EEZ.DIB.MCU.r1B1.zip)

Current lengths, 
Address Group:
A0-A5 - 45.0
A6-A11 - 45.5

Command Group:
SDNRAS - 45.0
SDNCAS - 53.7
SDNWE - 45.0

Control Group:
SDBA1 - 50.0
SDBA0 - 49.5
SDNBL0 - 50.0
SDBLN1 - 54.4
SDCKE0 - 45.0
SDNE0 - 47.0

Data Group:
D0-D1 - 45.0
D2 - 48.4
D3 - 48.4
D4-D12 - 45.0
D13-D15 - 50.5

Clock:
SDCLK - 55.0

prasimix, 2018-09-08:
Thanks, I believe there is no need to go down to +/-1mm. First, we have to see what will happen when I tried to run existing PCB (that should be in a week or two). 
Should I just add that changes into my next commit or you are going to initiate a new pull request?

Rerouter, 2018-09-08:
I'll Initiate, As I have also wrapped up Kicad-ising the schematic, and generating a netlist.

Rerouter, 2018-09-08:
https://github.com/eez-open/modular-psu/pull/5

Pull request made.

prasimix, 2018-09-09:
I managed to open KiCad file today. Unfortunately it seems that it doesn't include bug reported and fixed in Eagle under #3. Any chance to include that change?

Rerouter, 2018-09-09:
https://github.com/eez-open/modular-psu/pull/6

All kicad files are now brought up to eagle master.

Rerouter, 2018-09-09:
https://github.com/eez-open/modular-psu/pull/8

Missed an unconnected count,

prasimix, 2019-03-13:
Prototype works predictably, can be reopened if some issue arise.

closed 2019-03-13 by prasimix