eez-open/modular-psu #54: Proposal switch from STM32F7 to STM32H7 Dual Core

bvernoux, 2019-11-23:
I have seen selected MCU is an STM32F769IGT (LQFP176) which is very good anyway I think for future and to manage even more things it will be a must have to switch to pin compatible STM32H7 CPU with the big advantage to have Dual Core 1x CortexM4F @ 240MHz and 1x CortexM7@480MHz tons of FLASH/RAM peripherals ... that will allow to add special hard real-time monitoring / tasks... in one core and doing other things in an other core ...
It seems STM32H757IITx LQFP176 is full pin compatible (more check to be done to be sure) marketing status is active but it seems hard to find so far (maybe contacting ST directly will help)

bvernoux, 2020-02-04:
Just an update as now some STM32H757IIT6 are available at mouser EU (https://eu.mouser.com)

prasimix, 2020-06-23:
I invested some time to check out the H757. Unfortunately it is not a pin compatible with the F769 and needs a lot of changes. Furthermore, I failed to configure in CubeMX to have Ethernet, SDRAM and TFT display (RGB) at the same time. However, it could be relatively easy to migrate to H743.

bvernoux, 2020-06-23:
Thanks for update main interest in STM32H7 was mainly dual core (H7 vs F7 are better but less interesting I think) but yes it seems not easy ...
Other idea to keep Dual Core features is to check STM32H745I / STM32H747I or STM32H755I (with LQFP176) maybe they are "more pin compatible" to actual STM32F769IGT and can be configured with Ethernet, SDRAM and TFT display (RGB) at the same time...

prasimix, 2020-06-23:
I think that 176 pin package is simply too small to assign concurently what I'd like to have, and that is:

1. Ethernet MII or RMII
1. SDRAM min. 12 address lines
1. TFT RGB 16-bit
1. USB OTG/VBUS detection
1. SD card 4-bit data bus

bvernoux, 2020-06-24:
Maybe it will be required to switch to the "big" version TFBGA240 to have all possible features, I think it will be a very nice evolution for future (and mainly features/extension which requires hard real-time).

prasimix, 2020-06-26:
I would rather try to do something with the LQFP208. I have never designed a PCB with BGA before, and I am not equipped to assemble PCB with such a package. In the meantime I could spend some time trying to migrate to single core H7 like already mentioned H743 since it offers more speed for lower price then F769.

chenpeng198877, 2020-07-20:
Can stm32f769bit6 be used?

prasimix, 2020-07-20:
> Can stm32f769bit6 be used?

It is not possible with the current PCB revision, and the idea discussed here is to move to the MCU from the H7 family (preferably dual core).

prasimix, 2020-07-20:
BTW, I managed to layout a new PCB revision based on the STM32H743IIT6 (LQFP-176) which is still single core but offers higher speed for less money. Of course, we should make a prototype and support it in the firmware.

bvernoux, 2020-07-24:
Yes it is already a good step/compromise as dual core (is much more expensive/complex..) and requires lot of more works maybe for a future update ...

prasimix, 2023-11-20:
If we go with a new MCU, according to the latest research, it could be the H745 (dual-core).