eez-open/modular-psu #63: mcu KiCad files are obsolete

bvernoux, 2020-01-06:
mcu KiCad files are obsolete they use EEZ DIB MCU r1B2 when official latest version (Eagle files) is EEZ DIB MCU r1B6

It will be great add details in each directory that official design are done only on Eagle to avoid confusion and mistake if someone want to use/work with latest validated design files.

A must for a fully open source hardware (especially certified oshwa https://certification.oshwa.org/hr000002.html) will be to have all latest design files converted to fre/open source software and here especially the amazing KiCad

goran-mahovlic, 2020-01-06:
Maybe we can try to convert latest version with https://github.com/lachlanA/eagle-to-kicad
I am also not Eagle fan!

goran-mahovlic, 2020-01-06:
@bvernoux for test I have converted bp3c board so you can check, and tell if you have any suggestions - I will convert rest of them on my repository ...
https://github.com/goran-mahovlic/modular-psu/tree/master/bp3c/KicadFiles

goran-mahovlic, 2020-01-06:
Seams that new kicad knows to do conversion without scripts, so I will try that way

bvernoux, 2020-01-06:
Yes it is advised to do it directly with latest KiCad 5.1.5 (but anyway it often requires some manual fixes)