eez-open/modular-psu #65: Farnell and RS parts in 1-click-bom.csv are invalid

kasbah, 2020-02-20:
Just noticed there are a lot of parts numbers that are not correct for Farnell and RS for e.g. https://github.com/eez-open/modular-psu/blob/master/aux-ps/BOM/1-click-bom.csv.

Easiest thing to do would be to just remove those columns to avoid confusion.

goran-mahovlic, 2020-02-20:
Will check and fix tomorrow

goran-mahovlic, 2020-02-21:
I have recreated all BOMs so we can check after refresh on kitspace...

kasbah, 2020-02-21:
Nice one, I can still spot one error in the [aux-ps one](https://github.com/eez-open/modular-psu/blob/master/aux-ps/BOM/1-click-bom.csv) (line 3, C0805C105K4RACTU for RS). It's also in [this one](https://github.com/eez-open/modular-psu/blob/master/dcm220/BOM/1-click-bom.csv#L8) and [here](https://github.com/eez-open/modular-psu/blob/master/dcp405/BOM/1-click-bom.csv#L16). 

Can you avoid using a `-` in the Digikey column or do you need it for something? It will cause errors when trying to add to cart I think.

goran-mahovlic, 2020-02-21:
Tnx, please recheck now

kasbah, 2020-02-21:
Looks better, [line 35](https://github.com/eez-open/modular-psu/blob/master/aux-ps/BOM/1-click-bom.csv#L35) is moved over by one I think?

goran-mahovlic, 2020-02-21:
yes - now it is fixed.
I am using your 1 click BOM generator and autofill - and sometimes it adds problems

kasbah, 2020-02-21:
Hmmm... it really shouldn't move things over like that! :scream: 

Are you also doing some manual editing that might have introduced that mistake? If you can reproduce the problems somehow, please do report. https://github.com/kitspace/bom-builder/issues

goran-mahovlic, 2020-02-21:
Yes I am copy rows from here 

https://github.com/eez-open/modular-psu/blob/master/aux-ps/BOM/EEZ%20DIB%20AUX%20PS%20r3B3%20BOM.ods

And pasting to csv -
After that I do upload to generator -
Wait until it does search -
Click on Autofill -
Save

kasbah, 2020-02-21:
Hmm, do you have a copy of the csv you pasted to from the ods, before you upload to the BOM Builder?

goran-mahovlic, 2020-02-21:
will try to recreate and open issue if I catch error

https://github.com/kitspace/bom-builder/issues

goran-mahovlic, 2020-02-21:
Reason of shift by one it is here:

[image]

kasbah, 2020-02-21:
Hmm, still if the CSV doesn't use a `;` seperator, that shouldn't be a problem. Even then, if it's quoted should be fine too. 

New BB3 pages are now deployed on kitspace.org btw. Unless you see any problems, I guess this can be closed: :tada:

goran-mahovlic, 2020-02-21:
Yes, it is probably something that needs fix on BOM bilder ...

If you notice new issues with BB3 we will reopen this issue

closed 2020-02-21 by goran-mahovlic

kasbah, 2021-07-14:
Hey, in [dcm220/BOM/1-click-bom.csv lines 55-56](https://github.com/eez-open/modular-psu/blob/master/dcm220/BOM/1-click-bom.csv#L55) the MPN columns seem to shift the distributor columns over.