eez-open/modular-psu #78: dcp405 cc/cv switch mechanism

sfmth, 2020-07-21:
How do you control the cc/cv mode. I mean does the mcu decide which to enable or do they switch on their own

prasimix, 2020-07-21:
Connected load and set output voltage and current defined mode of operation. MCU role is only to set output params. For example if U_SET is 10 V and I_SET is 1 A, in accordance with Ohm's law, load with resistance of 10 Ohms or more sets CV mode, and for lower resistance it will switch to CC mode since it cannot deliver more then 1 A.

sfmth, 2020-07-21:
When the output voltage is 10v and current set for 1A, we connect a 20 ohm load. How is it that the current comtrol loop doesn't try to drive the main control transistors to get a 1A feedback

prasimix, 2020-07-21:
> When the output voltage is 10v and current set for 1A, we connect a 20 ohm load. How is it that the current comtrol loop doesn't try to drive the main control transistors to get a 1A feedback

Because output of DCP405 as any other available power source is not an ideal current source. It is not ideal voltage source either. Ideal current source has infinite open loop voltage. Therefore, if you set its output to 1 A it can provide it with any load, and for 20 Ohm it will deliver 20 V, and not 10 V as in our example.

sfmth, 2020-07-21:
U_set is 10v, I_set is 1A  , load is 20 ohms.
Current control loop will try to provide 1A and would set the output voltage to 20v. But the  voltage control loop is set to 10v and tries to drive the output down to 10v rather than 20v . Now we will have a fight between The voltage regulating and current regulating OpAmps. 
How did you prevent this in your design

prasimix, 2020-07-21:
There is no fight as can be proven with measurements. You have to take into account initial state and finite time to reach any of set values, and that CC loop is by design slower then CV loop. Therefore set voltage always takes precedence over set current. In our example if voltage is set to 10 V that is a max. final value under normal condition. Only in error condition, e.g. major fault or externally applied voltage source with more then 10 V and enough current capability may lead to an increase in voltage. In that case software or hardware OVP will trip and turn off DCP405 output (if they are enabled).

sfmth, 2020-07-21:
Oh, so that's why LTC2050HV has a bandwidth around 1kHz rather than 10kHz on TL072

closed 2020-07-23 by sfmth