eez-open/modular-psu #79: any comments on my project?

sfmth, 2020-08-01:
https://github.com/sfmth/heirs-cheap-diy-bench-power-supply

Im working on a cheap purely linear bench power supply I would be glad to hear your professional opinion on my project.

The readme.md was a readme at first but turned into a design idea diary

I don't have a power supply and they are kinda expensive in my country so i decided to make one

prasimix, 2020-08-02:
You have made the first good step, and it is the specification that will lead you to the desired goal. The real job is just waiting for you in a few years (or sooner) with enough effort and time you will probably get what is on the list. I wish you great success.
I guess you have some kind of oscilloscope on hand because it is an unavoidable tool in the more advanced stage of development and testing. Of course, it would be very useful to have some lab bench power supply, but yes, you just have to put it together :)

sfmth, 2020-08-03:
Thanks, unfortunately I don't have an oscilloscope, all I have is a 5$ multimeter. I'm gonna go with it, maybe make an oscilloscope when I needed one :)

closed 2020-08-04 by sfmth