eez-open/modular-psu #80: IC9B gain tolerance effect on the adc reading

sfmth, 2020-08-04:
IC9B has a variable gain and thus the V_MON can vary to 10 mV. How do you compensate for that

prasimix, 2020-08-04:
Variable gain, in which domain DC or AC? U_MON can vary even more up to 50 mV but for less then 50 us as measured with transient load tester (i.e. AC domain). That will not affect U_MON reading using ADC (which is effectively only 15-bit in best case) since its sampling rate is very slow (2 KSPS max.).

sfmth, 2020-08-05:
No, I mean the inaccuracies due to component tolerances  in the gain of the IC9B. Just 0.1 percent tolerance in the resistors can cause a variation of 10 mV between different products. Not in microseconds, the resistor values can't change that fast. Just between different products. And the U_MON can have 10mV tolarance, so when you multiply the readings of the adc by a preset value in software the readings wouldn't be accurate as the gain has tolerances and we are using a preset multiplier in software which can't compensate for the tolerance of the gain. I guess this would require calibration of each device but I didn't see that mentioned anywhere. Without calibration this would bring the accuracy down whilst the adc can do much better

prasimix, 2020-08-05:
Parts tolerances are not an issue if you can count on calibration. TCR is more important. This is especially evident on the Hi current range Rsense (R67) and this resistor is specially selected.

prasimix, 2020-08-05:
BB3 section in User manual about calibration: https://www.envox.hr/eez/eez-bench-box-3/bb3-user-manual/12-power-modules-calibration.html

sfmth, 2020-08-11:
[image]
Got my first oscilloscope today for 35$, you can't find a scope like this for less than 200$ here , this guy was kind and sold it for cheap to me. He has a electronics shop in the town

prasimix, 2020-08-11:
Great, that is a good step in right direction.

sfmth, 2020-08-15:
How did you filter the switching noise going to the regulator? I tried some filters but didn't remove it that well

prasimix, 2020-08-17:
Switching noise is not as simple to measure as it is advertised. You have to be well equipped (scope, probes, termination). Moving probe for just one millimeter could make huge difference. [Here](https://www.envox.hr/eez/eez-bench-box-3/eez-bb3-measurements/dcp405-power-module.html#dcp405_pard) you can find used measurement setup for power output PARD, and the same is applicable for measuring switcher power output. 
Filtering is required on both input and output side of the switcher and you can count on considerable attenuation with simple filters (e.g. output LC filter). Measurement has to be done on more or less finalized design i.e. soldered and wisely placed components on the PCB (forget about breadboards/protoboards). 
Even if you made reasonably silent switcher that is still not a guarantee at all that such design will pass EMI radiated emission testing (if you even came into position to certify your end product). On current power modules (DCP405 and DCM220) you can see input filters (against noisy Mean Well) and output LC filters.

sfmth, 2020-08-17:
[image]
Well this is the noise I'm trying to eliminate, I managed to get it 10 dB down with an active RC filter.
It has a big amplitude at 500khz, I don't think there would be that much of common mode noise, I'm running 2meters of wire from a fixed switching supply to my breadboard.
I'm trying hard not to use an inductor for the best transient response, how would I get a better filter without an inductor?

prasimix, 2020-08-18:
At what point was this measured? Switcher output after RC filter? Please explain "active" RC filter. Does it include op-amp or similar additional circuits? 
If you are using switcher as power pre-regulator, good transient response doesn't mean a lot, because final stage (linear post-regulator) will be in charge of good transient response.

sfmth, 2020-08-18:
This was measured at the tip of the cable going to the breadboard. It's a switching powersupply. I just want to make my design compatible with high input noise.
Here's the files
https://github.com/sfmth/heirs-cheap-diy-bench-power-supply/tree/master/Experiments/switching%20noise%20rejection%201.0

prasimix, 2020-08-19:
Forget for a while features of power source for your linear post-regulator stage. Make one step at the time: if you want to build post-regulator first, then make sure that it is powered from decently regulated source and make sure that it behaves correctly. First you have to achieve stability in all mode of operations, then you can start to dream about PSRR and other things.
When you got good post-regulator stage in next step you can start to think about your own pre-regulator, and if it is going to be a switcher, and want to make it silent enough you have to design it on the PCB. Think about 4-layer but you can start with 2-layer and difference could be significant.

sfmth, 2020-08-19:
Oh thanks a lot, I actually ordered some parts for the post regulator, I decided to work on this until they arrive. So I would focus on the post regulator and when finished I'm gonna go over other stuff. I actually wanted to make a separate switching pre regulator on a separate board and just make the pre regulator optional. Without the pre regulator the consumer gets less noise and less price.

prasimix, 2020-08-20:
Less noise? If you continue, you will quickly find out that there are a number of other problems and that the output noise will not be a priority. Beware, there is a lot of talk about low output noise, it's nice to dream of noise below e.g. 1 mVpp but that in itself doesn't mean much because the devices in this class come with a number of other hard-to-achieve features (long term precision and stability, fast response, lots of protections, etc.).
I believe you are aware that some kind of pre-regulation should exist and that it is not a fancy supplement but an inevitable thing (e.g. "entry level" pre-regulation is multi-tap transformer with relays). Therefore, I do not know what it would mean to offer a solution _with_ or _without_ pre-regulation. I would also recommend that you concentrate on a solution that will first satisfy all of your requirements that you will be able to prove to yourself over a period of time, and then possibly think about “customers”. If you set out to do this just for the sake of income/profit, there are other (and faster!) ways to do that. One of them would be to clone or organize the production of, say, EEZ H24005 :)
Don’t forget LTspice or similar simulators. They can significantly speed up wandering and finding an acceptable path before you spend a cent on buying components.

prasimix, 2020-08-24:
Use LTspice and it can give you many answers without wiring a single part on your breadboard.

sfmth, 2020-08-30:
Sorry if I answer this late, I was busy migrating to Arch Linux and doing a freelance job. Yes I know there should be at least transformer taps, I just wanted to make the switching buck converter pre-regulator optional. I'm planning on designing a novel & seamless analog switching system that switches between the taps. In the [PSL-3604](https://www.eevblog.com/forum/projects/diy-bench-power-supply-psl-3604/) there was a '4-level supply' used for switching between taps, in my calculations that system dissipates more power than a normal relay controlled tap switching system, whilst he claims that it provides less dissipation. I'm doing this project to go through the different processes of designing a product and gain experience. I used LTspice and it was great, you can follow my project in my [repository](https://github.com/sfmth/heirs-cheap-diy-bench-power-supply).

prasimix, 2020-08-30:
> I'm planning on designing a novel & seamless analog switching system that switches between the taps. In the [PSL-3604](https://www.eevblog.com/forum/projects/diy-bench-power-supply-psl-3604/) there [...]

Then it is possible that your calculation is wrong or you're started with false assumptions. Leonid's design offers reduced or equal dissipation compared with relay switched pre-regulation but also excellent dynamic that you can simply cannot achieve even with the fastest mechanical relays. For example, try to program output from e.g. +5 V that belongs to "level-1" or lowest secondary and then switch it to +20 V that requires level-2 or higher secondary. And try to do switch that up and down with 50 ms cycle, add e.g. load that draws few amps and check what will happened with relays :).
From other side Leonid's design has higher dissipation most of the time comparing with switching pre-regulator.

sfmth, 2020-08-30:
Well in his design all the main mosfets have the same Vgs so they can only pass the same current, as each one is connected to a tap and they are in series, only the highest tap would be  supplying the current. If other taps supply any current the current wouldn't be the same on all mosfets

sfmth, 2020-09-01:
I ran a [simulation](https://github.com/sfmth/heirs-cheap-diy-bench-power-supply/tree/master/Experiments/Tap%20switching%20topology) in LTspice, and it doesn't work. can you take a look and see what's wrong?

prasimix, 2020-09-01:
I would recommend you to ask things like this on the EEVblog [forum](https://www.eevblog.com/forum/beginners/). For BB3 related stuff you can open new issues in GitHub.

closed 2020-09-01 by prasimix