eez-open/modular-psu #87: Allow external current programming?

madscientist159, 2020-11-01:
Is there a way, using the existing hardware, to allow for external current programming using the same input as would normally be used for external voltage programming?  This would be useful to integrate the BB3 into certain types of automatic test systems.

Thanks!

prasimix, 2020-11-01:
Currently that is not possible, but it can be added in the future revisions in the same way as with external voltage programming by adding another SN74LVC1G3157 and using one output pin on the I/O expander (we have one spare). Since there is not more space available on the module front panel, external voltage and current programming should share the same terminals. Therefore only one of them could be active in any moment, what shouldn't be a problem.

madscientist159, 2020-11-01:
I have a spare module and some time to play with this...I assume there are also firmware changes needed?  If I just wanted to hack a module into current only mode, would I need to add the IC or could I just reroute a wire or two?

electrokean, 2020-11-02:
@madscientist159 If you are happy to hack the board, then you could add a physical SPDT switch for this.  Look at the schematic page 4.  There is a jumper labelled "I_SET" (just "I" on the PCB) next to IC16 which you could cut.  The add the switch to select between shorting the jumper or injecting an external signal into I_SET.  You could/should reuse the RProg+ input and protection components for the injected signal.  Of couse it would be better to add another SN74LVC1G3157 controlled by GPB5, but that will  require firmware changes.

electrokean, 2020-11-02:
Oh, one word of caution - the I_SET signal is also sampled by the ADC.
I'm not sure if the firmware will get upset if it sees an unexpected value due to the injected signal.
You can probably find this out pretty quickly by just trying it, or with a bit of time spent reviewing the code.

prasimix, 2020-11-02:
I'm not sure that any kind of hack would be possible without intervention in firmware, e.g. if you want to use existing switch (IC17) for current, you'll lost control of set voltage, it will be zero, and in that case externally programmed current does not make any effect.

electrokean, 2020-11-02:
@prasimix I wasn't suggesting removal of IC17, just a similar set up to control I_SET from the external input.

U_SET control can of course bypass IC17 via JP2 - hey, I can use that on my board missing that part until the replacement arrives.

My area of concern was what does the firmware use AIN1 (monitoring of I_SET) for?
Does a wildly different value from the DAC set value cause a fault trigger?

prasimix, 2020-11-02:
> @prasimix I wasn't suggesting removal of IC17, just a similar set up to control I_SET from the external input.
> 
> U_SET control can of course bypass IC17 via JP2 - hey, I can use that on my board missing that part until the replacement arrives.
> 

Yes, that's right you can bypass IC17, but just wanted to say if you want to cheat existing firmware to control switch for current instead of voltage that won't work.

> My area of concern was what does the firmware use AIN1 (monitoring of I_SET) for?
> Does a wildly different value from the DAC set value cause a fault trigger?

I think that is not a case now, it is used for calibration only, but it could be used for undercurrent detection when output is in CC mode.

fietser28, 2021-01-27:
This enhancement is included in proposed design in #94.

closed 2025-02-16 by prasimix