eez-open/modular-psu #106: Oscillation between CC and CV while charing batteries

willg, 2021-04-01:
This was discussed on the EEVblog [here](https://www.eevblog.com/forum/projects/eez-h25005-a-possible-successor-of-eez-h24005-programmable-power-supply/msg3411354/#msg3411354), and Discord [here](https://discord.com/channels/511131497801973761/511133610556784640/824318110218977370) but the BB3 will oscillate between CC and CV modes while charging a battery when the battery voltage gets within ~10% of the CV set point.  This occurs because Q27 is turned on when CC becomes active, and the CV set point drops with the addition of R118.  When charging a battery, this voltage drop the causes the battery charge current to drop below the CC set point, Q27 turns off, the voltage set point rises and the supply oscillates.  

I've used a variety of other supplies in this exact same way and I haven't observed oscillations like this.

prasimix, 2021-04-04:
This issue require firmware change as reported in [#177](https://github.com/eez-open/modular-psu-firmware/issues/177) and removing of Q27 in r2B11 version or Q10 in r3B3 version of the DCP module. That MOSFET was added against power up overshot on lower set output voltage. For example without MOSFET when output voltage is set to 2 V, output voltage without load look like this:

[image]

If small ramp is applied (2 ms) this issue vanished:

[image]

willg, 2021-04-06:
Removing Q27 and adding the 2ms ramp looks like a nice solution to me.  

I'm happy to modify one of my DCP405s and help test out the change once [issue #177](https://github.com/eez-open/modular-psu-firmware/issues/177) is implemented.  Thanks again for taking the time to work on this!

prasimix, 2021-04-11:
Ramp is needed on both power up and OE on if set current is below 170 mA.