eez-open/modular-psu #108: R4 overheats on the AUX-PS r3B3 module (green PCB)

prasimix, 2021-06-12:
R4 which limits the opto-triac current in the soft-start circuit overheats if BB3 is switched on continuously for several days. 

The soft-start (inrush current) limiter circuit was not new to the AUX-PS module: it was literally copied from the H24005 PSU in which it has been running properly for years. However, the H24005 does not use the Mean well LRS F-model (full range or autoswitch) but one that has manual input voltage selection (115/230 Vac). So basically it's another model! Soft start uses two triacs that are controlled via the opto-triacs OK1 and OK2. It turns out that the Mean Well F-model in some way affects the operation of the opto-triac even when there is no load. In the long run, i.e. if your BB3 is turned on continuously for a week or more, the power resistor that should limit the current of the opto-triac overheats so much that it eventually burns out.
Why haven't I been able to notice this so far? For two reasons: I have BB3 active for at least 10 hours a day every day, and when I did continuous testing (which lasts for several days) then I did it with BB3 which had one modification on Mean well's which was one of the attempts to passes the current harmonics test. So I had 4u7/X2 capacitors connected in parallel at their inputs! Here are some pictures.
The resistor that looks good to me but not to @goran-mahovlic is the R4 which is under increased stress from Mean Well's F-model:

[image]

R4 temperature if two F-models are connected without added capacitors at their AC inputs (230 Vac):

[image]

... and it increases further.
If I connect a purely resistive load (230 Vac, 46 W halogen lamp) R4 temperature is just fine:

[image]

The same is true if I have capacitors added:

[image]

Detail of the connected capacitor on one of the Mean well's:

[image]

prasimix, 2021-06-12:
I can't test 115 Vac and 60 Hz here, I only have 50 Hz, but it seems that the situation is even more alarming. Fortunately, capacitors help in this case as well. Without capacitors (115 Vac / 50Hz):

[image]

... with added capacitors:

[image]

prasimix, 2021-06-12:
What will happen if R4 burns out? Soon after the power up DCP and DCM modules will report an error as they will get DC at their inputs only for a short time (as long as the soft-start sequence lasts when another triac will be used).
In short, no further damage should be expected.

goran-mahovlic, 2021-06-12:
https://youtu.be/KNH0OdQx7IY

johnsonm, 2021-06-12:
In the picture with the capacitor, do you also have another small component? Resistor or diode? At least on my screen, I can't tell...

goran-mahovlic, 2021-06-12:
@johnsonm I have asked the same - it is nothing important just resistor for quicker capacitors discharge after power down

prasimix, 2021-06-13:
The capacitor used is 4u7, X2 Kemet [R46KR447050P1M](https://hr.mouser.com/ProductDetail/80-R46KR447040P1M). A bleeder resistor can be connected in parallel (470K to 1Meg):

[image]

prasimix, 2021-06-13:
It seems that only one capacitor is needed, even if three Mean well's are used:

[image]

goran-mahovlic, 2021-06-26:
I could only find 0.47uF on some old power supply

[image]

After adding it many problems just dispersed:

Resistor R4 is not overheating any more!
With function generator I could repeatably get smoke out of DCP405 with setting voltage to 3.3V and current to 100mA with this capacitor it is working normally!
AUX OTP tripped after long usage is gone
I could not use function generator if OVP HW is set - it was activated in first few seconds, now it does not
Current showing on channel without load was much more noisy
 
I have two channel BB3 with two mean wells.

ochococo, 2021-07-01:
I have two DCP405. Powered up for months now (usually around 15W per channel, rarely higher). Zero issues. 
Today I've added two `1uF WIMA MKS4 10% 400V` just to sleep better 🙂 

Thanks everyone for sharing and @prasimix for transparency!

gregdavill, 2021-07-04:
I'd not known about this issue. But just today after having my unit powered for 10 minutes I bumped the table and I get a CH1-Ch2 Error message. I'm in Australia so running at 230V,50Hz.

Opening my unit I saw this.
[image]

And I can confirm that it's R4. It appears to gotten hot enough to melt the solder, and then after a slight bump it simply fell off.
[image]

I could do some tests later with a thermal camera this week if you want some more data from systems running at 230/240V.

xlfe, 2021-07-04:
Weird - I'm in Australia, and after 10/15 minutes not seeing any significant heat on R4[
[image]
[image]
](url)

gregdavill, 2021-07-04:
I must just have the worst luck with my unit then. :laughing: 

I've gone ahead and installed 2x 1uF X2 caps. 
[image]
[image]

While it helps... the resistor still reached ~88.0C after 5m of idle operation. (Measured with a kapton insulated thermocouple.)

I also took a short video with a thermal camera.
https://youtu.be/8bMoKZcSGj8

Unfortunately I don't own a radiometric unit, so I can't read off exact temperatures, only the thermal flux. The modified version I took a reading with the thermocouple right after capturing this video, which read 88.0C. 

You can see the modified version does a _bit_ better than unmodified.

My local electronics store didn't have 4.7uF X2 caps. But I can order some with my next Digikey/Mouser order. Any other ideas about why this is happening? Or things I should try.

xlfe, 2021-07-04:
Hmm yeah that's strange. Not sure if it's relevant but my unit has three meanwell LRS-150-48s instead of two. I have a few R46KW447000M1K on the way. Happy to post one to you when they've arrived

prasimix, 2021-07-04:
@gregdavill this is really strange since @goran-mahovlic reported that only single 0.47uF rectify this problem for him. I tested it on my side with 4.7uF as already reported. I still don't know why combination of two Mean Well's makes this problem. For the test could you please try to disconnect one Mean well and make thermo measurement once again? Thank you very much for your patience.

prasimix, 2021-07-04:
I tried right now two new Mean Well randomly selected from piles and another AUX-PS "green" version. Not the slightest change in temperature can be measured on R4. See pictures (I used a wooden stick to make it easier to see on the thermal image where the R4 is). Still don't know why is that.

[image]

[image]

prasimix, 2021-07-04:
Ok, obviously something is wrong with the combination of multiple Mean Wells. But not only the two LRS-150F but also the two IRM-10s located on the AUX-PS module. I did a new test and instead of just a capacitor I put an RC snubber 1uF + 47R (3-5W) on the AC input and now I have a significantly better situation. It seems that a little experimentation will be needed here. 

[image]

gregdavill, 2021-07-04:
I did repeat the test here with just one meanwell LRS-150F unit attached. I received the same result, temperature rapidly approached >80C within a few minutes.

prasimix, 2021-07-04:
Ok, as you can see I got better results with bigger RC snubber (since one exists on the AUX-PS module), it now looks like this:

[image]

gregdavill, 2021-07-04:
Adding a 1uF / 47R to one unit does appear to help me a bit. This is brings my results inline with your initial testing, R4 climbs to around ~45C.

xlfe, 2021-07-04:
> I tried right now two new Mean Well randomly selected from piles and another AUX-PS "green" version. Not the slightest change in temperature can be measured on R4. See pictures (I used a wooden [...]

So what happens if you swap the AUX-PS with the AUX-PS on the other unit where you saw the high temperature rise? I can you isolate whether it is one or two dodgy LRS on your original unit, or a variation in the AUX-PS module?

prasimix, 2021-07-04:
I tried to swap LRS modules, but not AUX-PS (where IRM-10s resides). I'll do that and report here.

xlfe, 2021-07-04:
Okay

I guess what I meant was

| AUX-PS module  | LRS modules | Problem evident? |
| ------------- | ------------- | -- |
| A  | 1,2  | Yes |
| B  | 3,4  | No |
| A | 3,4 | ?? |
| B | 1,2 | ?? |

gregdavill, 2021-07-04:
I have very little experience with triac/AC circuits, but I've been poking around with my unit here. 

I added the extra snubber circuit on both LRS-150F's. And removed R1/R3. Is it my understanding these simply aid in noise immunity for the gate? (But at the expense of extra current draw?)
With these two mods my unit seems to function more correctly, with R4 only warming to ~28C (18C ambient) after 5m.

Not sure about the negative ramifications to this mod, but thought I'd share as an extra data-point. 
[image]

thomasrussellmurphy, 2021-07-04:
For those of us without the hardware, where is the revision of this board source that is affected by this issue and same for the wiring diagram to the updated PSU modules?

prasimix, 2021-07-04:
> For those of us without the hardware, where is the revision of this board source that is affected by this issue and same for the wiring diagram to the updated PSU modules?

* Affected AUX-PS module (Crowd Supply kit version): https://github.com/eez-open/modular-psu/tree/master/previous%20designs/aux-ps%20r3B3%20(Crowd%20Supply%20edition)
* Latest version (CE approved): https://github.com/eez-open/modular-psu/tree/master/aux-ps

PyroDevil, 2021-07-04:
I have the green AUX-PS with three LRS-150F. I measured with the temperature probe of my multi meter, and it seem to be around ~37C (25C ambient) after 5 minutes with a load of 24V and ~13W and no modifications.

With a 1uF X2 capacitor and a 1M bypass resistor on one mean well it got down to ~35C with two ~33C. So I don't think I could reproduce this issue really.

prasimix, 2021-07-04:
@PyroDevil as already mentioned, this issue seems to affects kit versions with 2 Mean wells in the first place. So you're safe (i.e. with original RC snubber on the AUX-PS module), but you can add a bigger one if you like.

thomasrussellmurphy, 2021-07-04:
Some debugging thoughts:

- What is the current drawn on AC_OUT_L by all downstream components? A modest bandwidth current clamp to an oscilloscope required. Could this fail to meet Q2's latch/hold currents while still maintaining active downstream loads?
- What is the line input waveform at the LRS-150F terminals? Does any distortion correspond to discontinuities in the current measured?
- Do the loads (IRM + LRS) on all the AC/DC supplies attached to AC_OUT_L contribute to R4 getting stuck hot?
- (highly complicated) Would an isolated zero-crossing detect and driving PWR_DIRECT at each ZC help identify if Q2 isn't switching reliably?

prasimix, 2021-07-04:
Thanks Thomas for all your valuable suggestions. It seems that some more detailed investigation is needed. At the moment I think we have a good solution that does not require any intervention on the AUX-PS module which is already retired and will no longer be offered. Also in the future Mean Well LRS will be replaced with EPP model.

electrokean, 2021-07-04:
As I mentioned on Discord, I was concerned that the issue with R4 is due to the lack of a snubber across triac Q2 terminals.  See the MOC3052M datasheet Figure 8 for an example snubber configuration.  The snubber across AC_OUT_L and AC_OUT_N (or on the PSU) won't help as much as one across the triac terminals.  Another option is to try the snubberless version of the BTA26.  As the issue seems to only affect some units & configurations, it is hard to know whether that will be of help.

It is possible that the latch/hold currents of the triacs is the issue when combined with the lack of snubber, as they could be marginal especially at low ambient temperature.  At idle, I measured a single LRS-150 PSU (-F or non -F) draws 45mA RMS at 110V 60Hz, and 86mA RMS at 240V 50Hz.  In both cases the power factors were really low.  With the -F PSU PF was around 0.010 at 110V and 0.022 at 240V.  With the non-F PSU, the PF was 0.046 at 110V and 0.025 at 240V.

I have still not seen this on either of my BB3 units even with the problematic configuration of two -F PSU's at 240V, so I haven't been able to investigate/test this myself.  I'm certainly no expert having only a few low power triac based designs under my belt.

gojimmypi, 2021-07-04:
fwiw, I have the Crowd Supply version of the BB3: Firmware v1.7 with the green R2B4. I have 3x Meanwell LRS-150F-48 units powering 2x DCP405 and 1x DCM220.  I'm in the USA, 120V, 60Hz. No snubber mod. No load DCP405 both set to 5V output, and R4 still running cool. 

I don't have the flir camera, but I unplugged the unit and used the finger test.

Looks like @gregdavill saw the overheating within a short period of time, but original issue seems to indicate the problem is seen after many hours of continuous on-time. 

@prasimix  How can I help test and see if my unit has this problem?

gojimmypi, 2021-07-04:
Also: I was just about to tweet how my firmware upgrade was successful and I did not see the overheating, when I noticed this GitHub issue mentions _R3B3, green PCB_ but in contrast the [instructions](https://www.envox.eu/eez-bench-box-3/bb3-user-manual/13-firmware-upgrade/#bb3_main_mcu_fw_win) note:

> If you got BB3 as a kit through a Crowd Supply crowdfunding campaign and the color of the MCU module PCB is green, choose r2B4 (the newer version of r3B3 has a PCB in blue)

I have the older, green PCB and had chosen the R2B4 at power up time after my upgrade from 1.4 earlier today.

So is the problem with r3B3 (blue) and is the subject of this issue incorrect? Or is it a problem with r2B4 (green) , or is there a mismatch between r3B3/r2B4  and PCB colors?

PyroDevil, 2021-07-04:
@gojimmypi: I have the same configuration as you, and @prasimix just confirmed that people with three Mean Wells do not have this issue, and don't need to fix anything.

gojimmypi, 2021-07-04:
@PyroDevil thanks for that! Good to know. Although I'm glad my three Mean Wells do not have the problem... are you saying just by removing the 3rd one connected to my DCM220 that I would expect to see the R4 overheat?

PyroDevil, 2021-07-04:
@gojimmypi: I guess so. At least that is how I understood @prasimix. I have not tested this though.

gojimmypi, 2021-07-04:
@prasimix I think the instructions may have a typo:  I have a Crowd Supply version BB3 with a _green_ PCB and confirmed there's a `r3B3` printed on my PCB:

[image]

In contrast, the [instructions](https://www.envox.eu/eez-bench-box-3/bb3-user-manual/13-firmware-upgrade/#bb3_main_mcu_fw_win) note:

> If you got BB3 as a kit through a Crowd Supply crowdfunding campaign and the color of the MCU module PCB is green, choose r2B4 (the newer version of r3B3 has a PCB in blue)

Or is there literally a different "version" of `r3B3` green vs `r3B3` blue?

Should I choose `r2B4` because I have a Crowd Supply Green PCB, or should I choose `r3B3` since that's what printed on my board. Any problems if I had picked the wrong one and powered on?

electrokean, 2021-07-04:
@gojimmypi In this case the r3B3 version relates to AUX-PS PCBA
For the firmware, it needs to know about the MCU PCB which for us original backers is r2B4.

Check this for the details of all the PCBA in Crowd Supply systems https://github.com/eez-open/modular-psu/tree/master/previous%20designs

gojimmypi, 2021-07-04:
@electrokean aha! Thank you for the clarification. I was looking at the wrong board, the power supply PCB. Indeed my MCU board is green, and r2B4.

[image]

PyroDevil, 2021-07-04:
Right, its not about the MCU board version. The green Crowdsupply AUX PS is R3B3, and the blue CE certificed AUX PS board is R3B4.

thomasrussellmurphy, 2021-07-04:
> Thanks Thomas for all your valuable suggestions. It seems that some more detailed investigation is needed. At the moment I think we have a good solution that does not require any intervention on the [...]

Very good. The dual relay implementation in the current version of the AUX-PS will very definitely skip the troubles with triacs!

gillham, 2021-11-10:
Is the two channel Crowd Supply unit safe to purchase at this point?  Mouser shows 26 “CS-BB3-04” (two channel full) in stock.  They show 39 of the DCM220 module (CS-BB3-05) also but list it as End of Life. 

Are the in stock units with Crowd Supply / Mouser known good or are they still selling known bad units?

I would like to get a BB3 but the current state of this issue and overall manufacturing has me somewhat concerned. I don’t really want to shell out nearly $1000 USD on something that includes numerous obsolete modules and known issues. 

On the other hand the fix looks fairly straightforward (though it isn’t completely clear from all the comments this is the full final fix).

The availability of the newer units (with CE certification, EPP power instead of MeanWell etc) is vague enough at this point I’m wondering if I should just buy the in stock units from Crowd Supply and plan of modding them or using the End of Life DCM220 to avoid the issue?

I think some clarity on the current situation would help or those units are likely to be in stock forever as few people are likely to want a unit with a known failure out of the box for that price.

prasimix, 2021-11-10:
The Mouser / CS inventory is unfortunately anything but clear (at least for me). For example, at the beginning of the year, 65 sets of new BB3 - Starter kit (CS-BB3-03) were delivered, which have CE certified modules (this includes a new version of the AUX-PS module). They haven’t appeared on stock so far (and small chances have sold them out instantly). They also received over 70 pcs of DCM224 (CS-BB3-08) which are not in stock: unless they are the DCM220 you mention.
Maybe those two-channel BB3s on stock are actually Starter kits (CS-BB3-03) and DCP405 modules (CS-BB3-06), most of which should also be new (blue PCBs).

Related to the problem reported here with the old AUX-PS module: yes, adding a larger RC snubber has been proven to solve the problem.

electrokean, 2021-11-20:
Not sure when they were added, but I see 57 pcs of CS-BB3-08 now in stock at Mouser.  If those are DCM224 (no datasheet or info listed) then the price seems a little high at USD $250 vs EUR €177 (~USD $200).  But they are out of stock at Envox store, and if they had 70 then people are buying them.

prasimix, 2021-11-20:
Yes, CS-BB3-08 is DCM224.

gillham, 2021-11-20:
I attempted to contact Crowd Supply about this using their contact form on November 10th and never got a response other than the initial email telling me a ticket was opened.
Perhaps CS-BB3-08 being suddenly in stock is related to my ticket, perhaps not.

Considering all of the other things I've ordered from them, you would think they would be more interested in trying to sell me a CS-BB3-03.  Especially if they have 65 sets laying around in inventory somewhere.  :)

jeremyherbert, 2022-02-04:
@prasimix could you please confirm my understanding of the solution below (hopefully it can serve as a summary for others).

1. Buy a suitable 1uF safety capacitor, [example here](https://au.mouser.com/ProductDetail/KEMET/R463R410040M1M?qs=xHpv68zP4qc9WCi2Ex39gA%3D%3D)
2. Buy a 5W axial 47 ohm resistor, [example here](https://au.mouser.com/ProductDetail/TE-Connectivity-Holsworthy/EP5WS47RJ?qs=Odivq8ja526Qnei8GovsRw%3D%3D)
3. Connect hot/live/active (brown wire) to resistor, then to capacitor. Connect the other capacitor lead directly to neutral (blue wire). Heatshrink as appropriate.

Note: only one RC snubber is needed, even if there are three separate power supplies.

prasimix, 2022-02-11:
Yes, that's right.

vtegroup, 2022-03-29:
Any recommendations for a 120V 60Hz grid. Capacitor and resistor value to use.

jeremyherbert, 2022-03-29:
I think the parts listed above should work fine, they are rated for well over 120V

vtegroup, 2022-03-29:
Just think out load, I am not concerned with the rated voltage of the capacitor, but more with the values of the capacitor and the bleeder resistor for all users 230/120v. Why not use a  RC in combination with MOV as it is possibly due to transient/peak voltage.

closed 2024-12-28 by prasimix