eez-open/modular-psu #109: DCP module faults after running on function generator

fietser28, 2021-06-26:
Running FW 1.7-RC1.

Running the function generator on Channel 1 (DCP405) I get a module fault. I tried this a couple of times it happens between a few seconds after starting and 2-3 minutes. The BB3 and other modules keep working. A software restart solves the issue:
[image]

The offending function generator setting:
[image]

The output of the module looks like (nothing strange is happening):
[image]

No load (except for scope probes) was attached. Down programmer was on, current range was best.
Trying this with a 100ms period I couldn't reproduce the cash within 15 minutes.
Ethernet connection was active.
DCP405 is a R2B11.

prasimix, 2021-06-26:
Ok, that is a clearly result of newly introduced protection [#202](https://github.com/eez-open/modular-psu-firmware/issues/202). You don't have enabled neither SW or HW OVP. 
What is the connected load?

prasimix, 2021-06-26:
Hm, if load not connected, and you didn't move probes that somehow ~3% overvoltage is detected. But signal on the scope looks fine. Please try to move lower level above 1 V, and see what will happen.

prasimix, 2021-06-26:
I managed to reproduce this on my side. It seems like a bug in #202.

fietser28, 2021-06-27:
I tried several variations: increasing/decreasing the lower limit, increasing/decreasing the amplitude, adding a capacitor and or resistor as load, but I can't find a consistent pattern when it happens more frequently. To really find it requires repeating (and writing down) but that might take some time. All without OVP.

On the scope the timing of the fault (output going to zero) always has the same timing pattern: around 5 ms after falling edge.

Looking at the code introduced in #202 I'm wondering if noise plays a role here: A single sample might be off and at lower voltages this is larger in % and more likely to trigger (at 1V, 3% equals 30mV)?