eez-open/modular-psu #114: OVP PCB Inductor in DCP405 up in smoke

xlfe, 2022-07-14:
Made the mistake of connecting a battery thinking I could charge it, noticed some smoke and disconnected everything. 

[image]

Can't see any other damage. The module passes power on checks, but doesn't seem to provide any output - I tried a simple 10v 10ma into a multi-meter and it reads 10v 10uA but no the multi-meter shows no voltage.

[image]

[image]

Any tips on next steps? should I clean up the PCB Inductor traces and do a more thorough test of individual components?

prasimix, 2022-07-15:
Could you please let me know what exactly you tried to do with battery. 
As a first step, check if fuse F1 is OK.

xlfe, 2022-07-15:
Battery voltage was about 26v and I connected it to the main output on the module (+ve to +ve, -ve to -ve), set the module output to 28v and 1A and then turned the output on. OVP tripped immediately. after a few seconds I noticed the smoke, and disconnected the battery.

Just checked - Fuse 1 is fine, Fuse 2 is blown.
Also checked R93 next to Q13, and getting less than 0.5 ohms resistance

prasimix, 2022-07-15:
Ok, replace F2 and check if you can get set voltage output. You did something that is not allowed and I have to add a notice about that in the User manual: Never change output enable to ON when battery is connected and HW OVP is enabled!  
What is happened: The voltage on the output was 26 V, you turned on the output that started from about 0 V. HW OVP detected huge difference in output voltage and immediately tripped. Battery is obviously still charged enough that triac drawn tens of amps but F1 was too slow to protect PCB inductor. Sorry for that, feel free to replace existing F1 and F2 with fast acting 5 A fuses.

prasimix, 2022-07-15:
>  Also checked R93 next to Q13, and getting less than 0.5 ohms resistance

That's fine, you can try to somehow bypass damaged PCB inductor.

xlfe, 2022-07-15:
ok thanks. Yes as soon as it happened I realized I should have thought more carefully about what I was trying to do!

What is the PCB inductor used for? Ie what's the effect of bypassing it?

prasimix, 2022-07-15:
> What is the PCB inductor used for? Ie what's the effect of bypassing it?

It is intended to limit max. current to preserve triac. We're talking here about a couple of hundreds of amps in microseconds. In most of the cases it should works just fine with inductor shorted.

prasimix, 2022-07-15:
The following warning has been added to the User manual ([Chapter 10.2](https://www.envox.eu/eez-bench-box-3/bb3-user-manual/10-channel-protections/)):

[image]

xlfe, 2022-07-17:
Thanks - bodge wire that replaces the PCB inductor seems to have brought it back to life! Phew. Thanks!

[image]

closed 2022-07-17 by xlfe

prasimix, 2022-10-06:
Remedy for this is suggested by @brianbienvenu in [#255](https://github.com/eez-open/modular-psu-firmware/issues/255) that is actually similar to [#9](https://github.com/eez-open/modular-psu-firmware/issues/9) which is changed now to P1 priority.