eez-open/modular-psu #122: Where are the AUX-PS r3B4 module resources?

pierantoniomerlino, 2024-12-28:
I recently bought the AUX-PS PCB module. It is the r3B4 (09/2020) version and I discovered that the layout and schematics is quite different from the r3B7.
I search in the github repo (tags, branches and history) but I didn't found any info regarding that version. The r3B3 and r3B5 are available.

Where I can found the resoures for the r3B4 version?

prasimix, 2024-12-28:
You can find the requested information in the attached zip file...

[AUX-PS r3B4 (09_2020).zip](https://github.com/user-attachments/files/18265875/AUX-PS.r3B4.09_2020.zip)

closed 2024-12-28 by prasimix

pierantoniomerlino, 2025-01-17:
I apologize to bother you again...
I've also the MCU board r3B3 (09/2020) and I see some discrepancies between the components names in the board and the schematics I got from the repo.
For example, it seems that the C3 in the schematic is C4 on the board... Could you please point me to the correct resources?

Thanks.

prasimix, 2025-01-17:
You can find the requested information in the attached zip file...

[MCU r3B3 (09_2020).zip](https://github.com/user-attachments/files/18459640/MCU.r3B3.09_2020.zip)